rene-aguirre/pywinusb #17: No devices reported on Windows 8.1 64-bit, Python 3.4 64-bit

mohd-akram, 2015-01-12:
I previously wrote some code that ran on Windows 7 32-bit without any issues. Today I decided to run the same code on Windows 8.1 64-bit, but the device is not found. I then tried the show_hids.py example and I simply get this output even though I have several devices connected:

    There's not any non system HID class device available

mohd-akram, 2015-01-12:
I've tested with Python 3.4 32-bit and it works fine, so it only affects 64-bit versions of Python.

rene-aguirre, 2015-03-21:
On a second thought, your device might be claimed by Windows 8.1 (shipped with OS level drivers). This has being the case for some input page devices before.

Can you let me know more information about your device, special enumeration descriptors (e.g. show_hids.py on Win7 x64)?

mohd-akram, 2015-03-21:
I don't have that device any more, although I tested another one now (a gamepad) and it was detected.

Btw, there is a bug in the new version. `WINFUNCTYPE` should be imported from ctypes in `wnd_hook_mixin.py`, not ctypes.wintypes.

rene-aguirre, 2015-03-21:
This had being fixed on github already. I just refreshed PyPi. Thanks for confirming about not finding new issues.

closed 2015-03-21 by rene-aguirre

Lavalu, 2015-04-18:
i have a device that show the same message:

There's not any non system HID class device available

i use win 8.1 64bit, python 3.4.0 64bit, and pywinusb 0.3.7 from github. I tried different settings on my usb device. The usb device is a  mikrocontroller board (PIC32MX Clicker) i tried different vendor and product ID, HID classes, subclasses and protocol types but i always get the same message.

if i add a  print( device_path )  in the  core.py  find_all_hid_devices()  i get i list of devices including the PIC32MX Clicker.

This is a pic of a part of the standard usb descriptor, if you need any other part or if i can test other settings please let me know.
[image]

rene-aguirre, 2015-04-20:
How about your HID report descriptor?

Check with usbview utility, you might find more details:
https://msdn.microsoft.com/en-us/library/windows/hardware/ff560019%28v=vs.85%29.aspx

Feel free to post the full descriptor (including device endpoints).

Lavalu, 2015-04-20:
thanks for answering, i check my hid device with usbView but i did not find any value that may cause a sort as "non-system class HID" of my device.
I can change any of the values of my HID USB Device but did not know where to start.

here is the output of usbView:

[Port2]  :  USB-Eingabegerät

Is Port User Connectable:         yes
Is Port Debug Capable:             no
Companion Port Number:          0
Companion Hub Symbolic Link Name: 
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         no
 USB 3.0:                         no

Device Power State:               PowerDeviceD3

       ---===>Device Information<===---
English product name: "HID Custom Demo"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: Full (is not SuperSpeed or higher capable)
Device Address:                      0x01
Open Pipes:                            2

          ===>Device Descriptor<===
bLength:                                0x12
bDescriptorType:                   0x01
bcdUSB:                                0x0200
bDeviceClass:                       0x00  -> This is an Interface Class Defined Device
bDeviceSubClass:                 0x00
bDeviceProtocol:                   0x00
bMaxPacketSize0:                0x40 = (64) Bytes
idVendor:                              0x0000
idProduct:                             0x0300
bcdDevice:                           0x0100
iManufacturer:                      0x01
     English (United States)  "Mikroelektronika"
iProduct:                              0x02
     English (United States)  "HID Custom Demo"
iSerialNumber:                     0x03
     English (United States)  "0x00000003"
bNumConfigurations:           0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:              0x05
bEndpointAddress:           0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                    0x03  -> Interrupt Transfer Type
wMaxPacketSize:             0x0040 = 0x40 bytes
bInterval:                           0x0A

          ===>Endpoint Descriptor<===
bLength:                            0x07
bDescriptorType:               0x05
bEndpointAddress:            0x01  -> Direction: OUT - EndpointID: 1
bmAttributes:                     0x03  -> Interrupt Transfer Type
wMaxPacketSize:              0x0040 = 0x40 bytes
bInterval:                           0x0A

       ---===>Full Configuration Descriptor<===---

          ===>Configuration Descriptor<===
bLength:                            0x09
bDescriptorType:               0x02
wTotalLength:                    0x0029  -> Validated
bNumInterfaces:                0x01
bConfigurationValue:         0x01
iConfiguration:                    0x04
     English (United States)  "HID Config desc string"
bmAttributes:                      0xE0  -> Self Powered
  -> Remote Wakeup
MaxPower:                          0x32 = 100 mA

          ===>Interface Descriptor<===
bLength:                           0x09
bDescriptorType:             0x04
bInterfaceNumber:           0x00
bAlternateSetting:            0x00
bNumEndpoints:              0x02
bInterfaceClass:               0x03  -> HID Interface Class
bInterfaceSubClass:        0x00
bInterfaceProtocol:          0x00
iInterface:                         0x05
     English (United States)  "HID Interface desc string"

          ===>HID Descriptor<===
bLength:                           0x09
bDescriptorType:              0x21
bcdHID:                            0x0101
bCountryCode:                 0x00
bNumDescriptors:            0x01
bDescriptorType:              0x22 (Report Descriptor)
wDescriptorLength:          0x0021

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:              0x05
bEndpointAddress:           0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                     0x03  -> Interrupt Transfer Type
wMaxPacketSize:              0x0040 = 0x40 bytes
bInterval:                            0x0A

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:              0x05
bEndpointAddress:           0x01  -> Direction: OUT - EndpointID: 1
bmAttributes:                     0x03  -> Interrupt Transfer Type
wMaxPacketSize:             0x0040 = 0x40 bytes
bInterval:                           0x0A

rene-aguirre, 2015-04-20:
This looks good, but there might be a problem in the 33 (0x21) bytes HID
report descriptor format.

If it is not obvious to you by looking to the firmware byte tables, best
tool for reviewing is a USB protocol analyzer hardware like the beagle.

If you don't have access to this, try a software based protocol analyzer,
like usblyzer, they have a one month trial.

In general OS HID descriptor parsers are really harsh on size items and
collections. So right now if you can take a quick look to your code make
sure:
a) The the size in bytes for you HID report descriptor matches perfectly
(no more no less)
b) That each report don't assign more bytes than expected, your limit here
(as stated in the endpoints) is 64 bytes.

Lavalu, 2015-04-20:
the size of the descriptors in the firmware matches exact and 33 bytes is the biggest one, any other is smaller. 

Config descriptor have:
18 bytes,
9 bytes,
9 bytes,
9 bytes,
7 bytes and
7 bytes
= 59 bytes

HID report descriptor:
= 33 bytes

is 64 bytes the limit for both Config and Report Descriptor ?

i am not that familar with the usb descriptors, so it is difficult to me to change it without mistakes.
can you provide a Config and Report Descriptor from any other device that may work fine.
Maby one with with exact byte order or a firmware of any ohter device?

by the way i tried pywinusb with a windows 7 64bit os and my device with the same result: no device available...
So it seems like the device causes the error.

Verschickt: Mo, 20 Apr 2015 8:08 pm
Betreff: Re: [pywinusb] No devices reported on Windows 8.1 64-bit, Python 3.4 64-bit (#17)

rene-aguirre, 2015-04-20:
In this case the 64 bytes limit applies to your input/ouput/feature reports.

Feel free to post the 33 bytes HID report descriptor, on Windows only
drivers have access to this descriptor.

Usually, unless you'd be really familiar with HID report descriptors (
http://www.usb.org/developers/hidpage/HID1_11.pdf) you'd use macros
provided by your firmware SDK. Reason is HID reports schema is binary with
bit field level fields.

The rough way to design a HID report is to use the USB SIG HID report tool,
it's a GUI, but honestly is far from easy to use, this would only help you
for simple reports.

I remember I saw some script based tools on the web, you'd describe our
reports in XML/Json form and you'd get a header from it, but if you are
into firmware, I'd rather recommend getting C macros once you'd feel
familiar with the reports.

Other important item is validating your binary descriptor, also I remember
seeing some Linux parsers before, a quick search on google gave me this:
http://eleccelerator.com/usbdescreqparser/

Lavalu, 2015-04-20:
here is the HID Descriptor from the device:

// HID report descriptor
const uint8_t USB_HID_ReportDesc[33] ={
     0x06, 0x00, 0xFF,       // Usage Page = 0xFF00 (Vendor Defined Page 1)
      0x09, 0x01,             // Usage (Vendor Usage 1)
      0xA1, 0x01,             // Collection (Application)
  // Input report
      0x19, 0x01,             // Usage Minimum
      0x29, 0x40,             // Usage Maximum
      0x15, 0x00,             // Logical Minimum (data bytes in the report may have minimum value = 0x00)
      0x26, 0xFF, 0x00,       // Logical Maximum (data bytes in the report may have maximum value = 0x00FF = unsigned 255)
      0x75, 0x08,             // Report Size: 8-bit field size
      0x95, 64,               // Report Count
      0x81, 0x02,             // Input (Data, Array, Abs)
  // Output report
      0x19, 0x01,             // Usage Minimum
      0x29, 0x40,             // Usage Maximum
      0x75, 0x08,             // Report Size: 8-bit field size
      0x95, 64,               // Report Count
      0x91, 0x02,             // Output (Data, Array, Abs)
      0xC0                   // End Collection
};

every value is one byte long, if two are in one line the build a word ( for example 0xA1, 0x01 = 0x01A1 ).
lines with // are comments.

thanks so much for your time. tomorrow i will try to find a nice tool to build some hid despcriptors from your links/google. 

Verschickt: Mo, 20 Apr 2015 9:58 pm

  
  
  
  
  
 http://eleccelerator.com/usbdescreqparser/

Lavalu, 2015-04-23:
After testing my device with different tools and programming tools, i can verify that the device works correctly and the descriptor did not cause the problem.

The Problem was caused by Python version 3.4.0 . An Upgrade to 3.4.3 solves the Problem, so now the MikroE PIC32MX Clicker is easy available on Windows 8.1 64bit by pywinusb.

Thanks a lot for support and sharing this library.

  
  
  
  
  
 http://eleccelerator.com/usbdescreqparser/