rene-aguirre/pywinusb #18: Error on show_hids.py

vitome, 2015-03-14:
Hi Rene,
I tried to install your library, but when trying to execute the test on the object, some errors appear.
This is the output
```
C:\Downloads\pywinusb-0.3.3\pywinusb-0.3.3\examples>python show_hids.py
Traceback (most recent call last):
  File "show_hids.py", line 19, in <module>
    hid.core.show_hids(output = output)
  File "C:\Python27\lib\site-packages\pywinusb\hid\core.py", line 1556, in show_hids
    all_hids = find_all_hid_devices()
  File "C:\Python27\lib\site-packages\pywinusb\hid\core.py", line 121, in find_all_hid_devices
    with winapi.DeviceInterfaceSetInfo(guid) as h_info:
  File "C:\Python27\lib\site-packages\pywinusb\hid\winapi.py", line 428, in __enter__
    return self.open()
  File "C:\Python27\lib\site-packages\pywinusb\hid\winapi.py", line 439, in open
    (DIGCF.PRESENT | DIGCF.DEVICEINTERFACE) )
ctypes.ArgumentError: argument 3: <type 'exceptions.TypeError'>: wrong type

C:\Downloads\pywinusb-0.3.3\pywinusb-0.3.3\examples>
```
I run python 2.7.9 on a Windows 7 machine 64bits... Any idea??

rene-aguirre, 2015-03-14:
Thanks for the report, is this from latest repo right? (as opposed to downloading PyPi snapshot

rene-aguirre, 2015-03-14:
The hint so far is 2.7.9 ctypes might be enforcing DWORD types when calling, I'll test on my Win7/64 machine next week with forced DWORD types on the DIGCF class.

vitome, 2015-03-15:
Hi René,
Many thanks for your prompt reply. It should be the latest repository:
```
C:\Downloads\pywinusb-0.3.3\pywinusb-0.3.3\examples>pip show ctypes
---
Name: ctypes
Version: 1.0.2
Location: c:\python27\lib\site-packages\ctypes-1.0.2-py2.7-win32.egg
Requires:

C:\Downloads\pywinusb-0.3.3\pywinusb-0.3.3\examples>pip show pywinusb
---
Name: pywinusb
Version: 0.3.3
Location: c:\python27\lib\site-packages\pywinusb-0.3.3-py2.7.egg
Requires:

C:\Downloads\pywinusb-0.3.3\pywinusb-0.3.3\examples>
```
Many thanks for the hint... I tried to change the types of the DIGF class (something like
```python
DEFAULT         = DWORD(0x00000001))
```
but now I have this other error
```python
File "C:\Python27\lib\site-packages\pywinusb\hid\winapi.py", line 439, in open
    (DIGCF.PRESENT | DIGCF.DEVICEINTERFACE) )
TypeError: unsupported operand type(s) for |: 'c_ulong' and 'c_ulong'
```
I've the feeling that you would have tried something different...

vitome, 2015-03-20:
Hi René,
were you able to figure out any change I could try? Many thanks

rene-aguirre, 2015-03-20:
I have python 3.4 on github, but I can see 3.3 is on PyPi, so you are actually no using latest.

I'll update PyPi.

rene-aguirre, 2015-03-20:
The issue can't be reproduced with latest code (0.3.4)

rene-aguirre, 2015-03-21:
Tested both Python 2.7.9 x64 and 32 bits with pyWinUsb 0.3.5

closed 2015-03-21 by rene-aguirre