rene-aguirre/pywinusb #19: Timing of input

prioslo, 2015-05-15:
Hi,

First of all, congratulations for this library. It works great!

I have the following issue: I am trying to record the input from a peripheric USB device, specifically a drum kit. What I need is basically to save the temporal information, i.e. the time-stamps of the events (drum beats, I don't really need sound information).  The drum kit has 6 different pads, and I got to the point where Python recognizes the input. However, I'm finding trouble to set a clock to record the timing (I' d like Python to detect when the drum is beat and save it in a csv).

My code runs as follows:

from time import sleep
from msvcrt import kbhit

import pywinusb.hid as hid

def sample_handler(data):
    print("Gure data: {0}".format(data))

import sys
if sys.version_info >= (3,):
        unicode = str
        raw_input = input
else:
        import codecs
        sys.stdout = codecs.getwriter('mbcs')(sys.stdout)
    
all_hids = hid.find_all_hid_devices()
device = all_hids[0]
device.open()  
device.set_raw_data_handler(sample_handler)
               
#device.close()

Thank you very much!

Paula

rene-aguirre, 2015-05-15:
Indeed, a raw handler as pointed out in your example can provide you the lowest latency. First you actually need to get started timestamping your data (e.g. data comes, and append in new queue as timestamp and data pairs).

But in your application I'm assuming you need to lower the timing jitter, some thoughts:

* My rough testing and some users reports about speed is that we can cope to similar to native OS speed, so I think python and the data processing is not adding that much overhead.

* Data is collected by a thread using overlapped reading (wait for even), but quickly posted into the device reading thread.

* The device reading thread calls your raw handler, but also would parse HID reports.

So first, just add a timestamp to your incoming data in your raw handler, if this is not enough, go ahead and test disabling the HID report handling, if this is still not enough then try to add the timestamping in the reading queue incoming data handling.

prioslo, 2015-05-20:
Thank you very much for your answer!
I finally found a solution to save the timing of the events as a list. I post it below in case it is useful for future users:

from time import sleep
from msvcrt import kbhit
import pywinusb.hid as hid

global tiempos
tiempos = list()

def sample_handler(data):
    import time
    tiempos.append(time.clock())
    print("HIT")

import sys
if sys.version_info >= (3,):
        unicode = str
        raw_input = input
else:
        import codecs
        sys.stdout = codecs.getwriter('mbcs')(sys.stdout)
     
all_hids = hid.find_all_hid_devices()
device = all_hids[0]
device.open()  

device.set_raw_data_handler(sample_handler)

device.close()

closed 2015-05-26 by rene-aguirre