rene-aguirre/pywinusb #20: pywinusb finds only first device

filin86, 2015-07-08:
in different scripts, function find_all_hid_devices() return only first device.
if in debug mode step into HidDevice.__init__() function, then its normally adds to result. but in that case when referring to a device an error occurs:

  Path:      \\?\hid#vid_3208&pid_4840#6&34655a94&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  Instance:  HID\VID_3208&PID_4840\6&34655A94&0&0000
  Port (ID): 28
 Port (str):USB\VID_3208&PID_4840\00000000002
Traceback (most recent call last):
  File "C:\Program Files (x86)\JetBrains\PyCharm 4.5.1\helpers\pydev\pydevd.py", line 2357, in <module>
    globals = debugger.run(setup['file'], None, None, is_module)
  File "C:\Program Files (x86)\JetBrains\PyCharm 4.5.1\helpers\pydev\pydevd.py", line 1777, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:\Program Files (x86)\JetBrains\PyCharm 4.5.1\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc) 
  File "D:/Downloads/pywinusb-master-0.3.7/pywinusb-master/examples/show_hids.py", line 19, in <module>
    hid.core.show_hids(output = output)
  File "D:\Python34\lib\site-packages\pywinusb\hid\core.py", line 1570, in show_hids
    dev.open()
  File "D:\Python34\lib\site-packages\pywinusb\hid\core.py", line 415, in open
    raise HIDError("Failure to get HID pre parsed data")
pywinusb.hid.helpers.HIDError: Failure to get HID pre parsed data

rene-aguirre, 2015-08-01:
The error is triggered by a Windows API failure to parse the HID descriptors, which might have a bad format in this case. Is this your device?

Fixing this issue looks like a good idea, any bad design product could be used at least in raw mode access, which is the way most people is only interested though, but honestly I'd not like to support broken products.

rene-aguirre, 2015-10-20:
@filin86 do you still think this is a PyWinUSB issue?

If so can you provide HID descriptor details?

rene-aguirre, 2015-10-28:
@filin86 I'm closing this issue, anyway, feel free to provide more details and I'll re-open it.

closed 2015-10-28 by rene-aguirre