rene-aguirre/pywinusb #22: Intermittent python crash when closing HID device

c1728p9, 2015-08-11:
Sometimes when closing a HID device python crashes and an error dialog box displaying "python.exe has stopped working" is displayed.

rene-aguirre, 2015-08-11:
@c1728p9 thanks for the patch!

The patch looks safe, but I'd like to know more details about the bug for my testing (I'll merge the patch but refresh PyPi once I do my own testing):
- Confirmed windows version (e.g. Windows 8, Windows 7 64 bits, etc).
- Python version (also if 32 or 64 bits)

c1728p9, 2015-08-11:
Hi @rene-aguirre 

I have seen this problem on Windows 7 64 bit on 2 different computers.  One was running 32 bit python version  2.7.8 and the other was running 64 bit python version 2.7.10.  @geky also saw this problem on a 64 bit windows 7 machine.  I haven't tested this on a 32 bit machine, but I would imagine it would have the same problem.

This problem is also very timing specific, at least for the hid devices we are using - mbed boards.  The boards receive a 64 byte command on the interrupt out endpoint and then send a response on the interrupt in endpoint.  The response on interrupt in comes 1-2ms after the write to the out endpoint.  If just a few ms of delay is added after WaitForSingleObject in InputReportReaderThread there is enough time for the read to complete and the problem goes away.  Adding a debugging print statements here was enough of a delay to make the errors go away.  The crash also goes away if the buffers used by winapi.ReadFile are added to a global list to prevent them from being freed after the call (although this causes the memory to never be freed).

Me and @geky saw this problem when running pyOCD tests which try to stress the system by repeatedly opening a hid device, performing a series of reads/writes and then closing the hid device.

If you want to reproduce this problem it would probably be easiest to do with a HID device that continuously sends data to the pc on the interrupt in endpoint.  If a packet is sent while the read is in progress but the buffers have been released then the crash will occur.

Let me know if you have any more questions or need more details.  Thanks for the quick response on this issue.

rene-aguirre, 2015-08-23:
Re-open due pending side effects verification.

rene-aguirre, 2015-10-20:
Re-merged after regression fixes.

PR #23 (2015-08-11): Fix crash when closing hid devie and more
Fix #22 , along with other modifications.
closed 2015-10-20 by rene-aguirre