rene-aguirre/pywinusb #24: Inconsistency on buffer size ?

Overdrivr, 2015-08-11:
When trying to send an 8 byte buffer, I get the following error:

```shell
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Python34\lib\tkinter\__init__.py", line 1533, in __call__
    return self.func(*args)
  File "C:\Users\B48923\Documents\Testchip\04_Software\Client\GUI_Frames\Frame_manual.py", line 35, in send_frame
    self.model.send_frame(data,self.combobox.current())
  File "C:\Users\B48923\Documents\Testchip\04_Software\Client\Model.py", line 32, in send_frame
    self.usb_interface.send_frame(frame)
  File "C:\Users\B48923\Documents\Testchip\04_Software\Client\USB_HID_interface.py", line 49, in send_frame
    report[0].set_raw_data(buffer)
  File "C:\Python34\lib\site-packages\pywinusb\hid\core.py", line 1294, in set_raw_data
    % self.__raw_report_size )
pywinusb.hid.helpers.HIDError: Report size has to be 9 elements (bytes)
```

Why does the report is supposed to be 9 bytes ? I thought the standard was 8 bytes, such as indicated in ARM mbed cookbook :
https://developer.mbed.org/cookbook/USBHID-bindings-
Is there a way to change the report size ? Tried to search trough the code but couldn't find a simple way

rene-aguirre, 2015-08-24:
The buffer sizes are affected by the HID report descriptors, can you confirm that in your firmware?

Input, output and feature reports can be described in a separate way, so might not match the size, as of usage schema.

Feel free to post information about your descriptors here.

rene-aguirre, 2015-10-20:
Not a clear signal this is a library issue.

closed 2015-10-20 by rene-aguirre