rene-aguirre/pywinusb #26: read keystrokes from keyboard

bibekchitrakar, 2015-09-03:
How can I reuse the code in raw_data.py to read the keystrokes from a usb keyboard.
I am lost in the codes.

            device = all_hids[int_option-1]
            try:
                device.open()

                #set custom raw data handler
                device.set_raw_data_handler(sample_handler)

                print("\nWaiting for data...\nPress any (system keyboard) key to stop...")
                while not kbhit() and device.is_plugged():
                    #just keep the device opened to receive events
                    # how to read events here
                    sleep(0.5)
                return
            finally:
                device.close()

rene-aguirre, 2015-09-03:
HID input devices are "owned" by the OS.

You won't be able to, this is a Windows limitation (input devices get protected).

You can use standard msvcrt library calls getch() and getche() to get actual key codes.

Some keyboard expose non input device page usages like LEDs and Music controls. I think long time ago (Win XP/7) these pages were exposed as long as the descriptors provided separate collections for input page and LED/others pages.

Check what you have exposed with the show_hids.py example. Only make sure to remove any special drivers your keyboard might required to be installed in order to enable special keys.

rene-aguirre, 2015-10-20:
As is. Windows prevents from capturing system HID input devices.

closed 2015-10-20 by rene-aguirre