rene-aguirre/pywinusb #27: Ability to exclusively open HID device

c1728p9, 2015-09-10:
Currently pywinusb opens hid devices as shared.  This allows multiple processes to open a handle to the same device.  This is problematic for devices that need exclusive access.  It would be great if there was an option to open a HID device with exclusive access rather than shared access.

rene-aguirre, 2015-09-18:
Russ,

About this, I think that it would even make sense to make it the default.

But this is one good candidate to provide a new parameter with the new
exclusive access mode (a boolean flag maybe?)

Best regards

Rene

c1728p9, 2015-09-21:
I could definitely make locking the default behavior.  I was just unsure if anything relied on opening it as a shared device.  A boolean flag sounds good to me.

rene-aguirre, 2015-10-20:
@c1728p9 Can we close this issue already?

c1728p9, 2015-10-20:
Yep

closed 2015-10-20 by c1728p9

MicroWizard, 2015-10-26:
Hi!

It is a good solution to open devices with sharing enabled for enumerating and exclusive for real using.
I had to make these changes for two month age and works fine.

But I have to ask you (since there is the only forum where the issue could be know) that Win8.1 and Win10 tends to enable opening HID devices "shared" only at boot and after restart it enables "exclusive" mode as well.

It is a strange behaviour because as an application developer I have to avoid shared open. Is it possible to force Windows to make a device only exclusively accessible?

Thank you for any thoughts.

c1728p9, 2015-10-27:
I'm using windows 8 and I haven't seen this problem.  Can you provide more details, like which usb device, or how long after boot this behavior starts occurring?  I know that devices that are opened as shared by one process cannot be opened for exclusive access by another at the same time.  It is possible that this is happening, and a restart kills the process with the open shared handle?

MicroWizard, 2015-10-28:
The device is a custom device (composite device with 1 CDC and 5 HID channels). It uses TI MSP430 MCU with TI's USB stack. Works well with Win7. Nowadays while some of our client/colleagues upgraded to Win8/Win10 are we able to detect this behaviour. When _boot_time_ the device is attached it is installed as shared. After restart it is installed as exclusive. First case our software detects it and refused to run (for sure).
To make it clear: it happens after clean boot and clean restart. Only Windows does what is want to do.

Very strange.