rene-aguirre/pywinusb #30: Win10, amd64 : 'OverflowError'>: int too long to convert

guenter1952, 2016-02-11:
Hi,
after a few days I give up:
**Python 3.5.1** (v3.5.1:37a07cee5969, Dec  6 2015, 01:54:25) [**MSC v.1900 64 bit** (AMD64)] on win32
on two machines with **Win10** Prof. . python installed in c\python35, not in a program dir.
pywinusb==0.4.0 installed ( says: pip freeze)
With Eclipse release 4.5.1 (on one machine) an : `import pywinusb.hid as hid` results in: unresolved module
Only after entering `pywinusb` to the 'Forced Builtins' it is found by Eclipse.
Running the example **raw_data.py** brings on both PC's (with and without Eclipse):

```
Traceback (most recent call last):
  File "C:\Python35\lib\site-packages\pywinusb\hid\core.py", line 121, in find_all_hid_devices
    for interface_data in winapi.enum_device_interfaces(h_info, guid):
  File "C:\Python35\lib\site-packages\pywinusb\hid\winapi.py", line 466, in enum_device_interfaces
    byref(dev_interface_data) ):
ctypes.ArgumentError: argument 1: <class 'OverflowError'>: int too long to convert

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Ich\Eclipse_Workspace_3.5\MagScan\raw_data.py", line 65, in <module>
    raw_test()
  File "C:\Users\Ich\Eclipse_Workspace_3.5\MagScan\raw_data.py", line 19, in raw_test
    all_hids = hid.find_all_hid_devices()
  File "C:\Python35\lib\site-packages\pywinusb\hid\core.py", line 152, in find_all_hid_devices
    results.append(hid_device)
  File "C:\Python35\lib\site-packages\pywinusb\hid\winapi.py", line 445, in __exit__
    self.close()
  File "C:\Python35\lib\site-packages\pywinusb\hid\winapi.py", line 451, in close
    setup_api.SetupDiDestroyDeviceInfoList(self.h_info)
ctypes.ArgumentError: argument 1: <class 'OverflowError'>: int too long to convert

```
I could not find any help/solution for this issue.
All other examples copied from the web show up with the same problem. No one submits info on 32 or 64 bit -machine he is using, because I thougt, that it could be a 64-bit-problem.
Guenter

c1728p9, 2016-02-11:
I added support for testing with tox which runs examples\show_hids.py to test the following configurations:
Python2.7.10_32
Python2.7.10_64
Python3.4.3_32
Python3.4.3_64

This worked on the commit d0745bcfb948e51c4ed9a485b7a118ed742b69ea on all configurations.  It was a Windows 8 machine that I tested on.  Have you tried running on an older version of windows?

guenter1952, 2016-02-12:
...but the question is: Was it Windows8 32 bit or 64bit ?
I do have only WinXp 32bit and Win10 64bit machines.
What I have found so far:
winapi:
line 451: `setup_api.SetupDiDestroyDeviceInfoList(self.h_info) `  This seems to throw the exception in the compiler exec.

with `h_info`'s like: int: 1518254951824, 3030873844784 or 2099881957920

(... and I only wanted to request a HID device....) ;-)

rene-aguirre, 2016-02-12:
@guenter1952 thanks for the report, the issue seems to be ctypes not detecting properly the SetupDiDestroDeviceInfoList argument type, so we'd need to try using an explicit prototype,

can you add this code below line 60, this line:

```
SetupDiDestroyDeviceInfoList    = setup_api.SetupDiDestroyDeviceInfoList
```

so it looks like:

```
SetupDiDestroyDeviceInfoList = setup_api.SetupDiDestroyDeviceInfoList
SetupDiDestroyDeviceInfoList.restype = BOOL
SetupDiDestroyDeviceInfoList.argtypes = [
    HANDLE, # __in       HDEVINFO DeviceInfoSet,
    ]
```

Then in what is current line 452 (or 451 as reported in your exception), let's remove the setup_api prefix, so the code looks now like:

```
        if self.h_info and self.h_info != INVALID_HANDLE_VALUE:
            # clean up
            SetupDiDestroyDeviceInfoList(self.h_info)
```

guenter1952, 2016-02-13:
@rene-aguirre: Thank You for your help.
I am one step further...
The second exception is gone, the first one at  `byref(dev_interface_data) )` is still there:

```
Traceback (most recent call last):
  File "C:\Users\Ich\Eclipse_Workspace_3.5\MagScan\raw_data.py", line 68, in <module>
    raw_test()
  File "C:\Users\Ich\Eclipse_Workspace_3.5\MagScan\raw_data.py", line 22, in raw_test
    all_hids = hid.find_all_hid_devices()
  File "C:\Python35\lib\site-packages\pywinusb\hid\core.py", line 121, in find_all_hid_devices
    for interface_data in winapi.enum_device_interfaces(h_info, guid):
  File "C:\Python35\lib\site-packages\pywinusb\hid\winapi.py", line 477, in enum_device_interfaces
    byref(dev_interface_data) ):
ctypes.ArgumentError: argument 1: <class 'OverflowError'>: int too long to convert
```

rene-aguirre, 2016-02-13:
Well let's remove the last piece of ctypes automatic args detection, at the top SetupDiEnumDeviceInterfaces assignment, same, let's add explicit argument types: 

```
SetupDiEnumDeviceInterfaces  = setup_api.SetupDiEnumDeviceInterfaces
SetupDiEnumDeviceInterfaces.restype = BOOL
SetupDiEnumDeviceInterfaces.argtypes = [
    HANDLE,                     # _In_ HDEVINFO DeviceInfoSet,
    POINTER(SP_DEVINFO_DATA),   # _In_opt_ PSP_DEVINFO_DATA DeviceInfoData,
    POINTER(GUID),              # _In_ const GUIDi *InterfaceClassGuid,
    DWORD,                      # _In_ DWORD MemberIndex,
    POINTER(SP_DEVICE_INTERFACE_DATA), # _Out_ PSP_DEVICE_INTERFACE_DATA DeviceInterfaceData
    ]
```

Then remove the `setup_api.` prefix in the function call above your line 477 to SetupDiEnumDeviceInterfaces.

If you have troubles with this I can push a test branch, but I expect the changes to be relatively safe, so once you confirm I'd rather push the patch to master after my own testing.

guenter1952, 2016-02-13:
Fine, so far. No troubles with the corrections, .... exept:
```
Traceback (most recent call last):
  File "C:\Users\Ich\Eclipse_Workspace_3.5\MagScan\raw_data.py", line 15, in <module>
    import pywinusb.hid as hid
  File "C:\Python35\lib\site-packages\pywinusb\hid\__init__.py", line 30, in <module>
    from . import core 
  File "C:\Python35\lib\site-packages\pywinusb\hid\core.py", line 25, in <module>
    from . import winapi
  File "C:\Python35\lib\site-packages\pywinusb\hid\winapi.py", line 74, in <module>
    POINTER(SP_DEVINFO_DATA),   # _In_opt_ PSP_DEVINFO_DATA DeviceInfoData,
NameError: name 'SP_DEVINFO_DATA' is not defined
```
? but I see the 
`class SP_DEVINFO_DATA(Structure):` definition somewhat later.

kind regards
Guenter

rene-aguirre, 2016-02-13:
Let me rather push to a test branch.

rene-aguirre, 2016-02-13:
Feel free to test the branch test_issue_30

@c1728p9 Rus, if you already have the test bench ready, could you please validate the patch too (I only checked the patch solves the issue on python 3.5.1, Win 10, 64 bits)

guenter1952, 2016-02-13:
Alors, Ca fonctionne!
Now it is running thru, I did not get any data out of my USB device (Voltage Measurement, it needs configuration(1)) and a keyboard hit did not stop the program, but thats another story.
Thanks a lot

closed 2016-02-13 by guenter1952

c1728p9, 2016-02-14:
Yep, still have my test env ready.  This patch passes all the tox tests (with the tweak in #31 to support the latest version of pylint).

So it appears that this issue was introduced in python 3.5.0.  Without this patch, python 3.4.3 still works for both 32 and 64 bit builds.  I see the same overflow error when running 3.5.0 64 bit though.

Patch looks good to me!