rene-aguirre/pywinusb #32: can you get the information of the displays I plugin?

maicss, 2016-03-26:
I run your pywinusb, it can list only usb device I plugin(but just the keyboard and mouse, others like Bluetooth, cannot figure out. but i think it's already awesome. thanks for your code.), can you help me with that? thanks.

rene-aguirre, 2016-03-26:
The Windows OS restrict USB access to input page HID class devices.

The only way to avoid this is to write a filter driver, this is something I'd consider of bad technical choice, the reason is that Windows prevents access to such devices because it already exposes standard APIs for them.

You can receive standard mouse or keyboard events in any Windows app, as long as it provides an event loop, look for helper libraries, most are based on PyWin32 library, check for examples where windows events are managed, I did something related in the PnP examples.

Possible the only disadvantage is that you need a window (a valid window handle) that's the reason the PnP examples create a Window.

Now on Bluetooth... if you expose valid HID profiles or HID over GATT (http://windows.microsoft.com/en-us/windows-10/supported-bluetooth-profiles-mobile) Windows will map the GATT profile as standard HID devices.

As PyWinUSB looks for HID device class devices, in theory you could interact with Bluetooth HID/HID over GATT without any special driver :-)

Provided your device doesn't use any input page usages.

rene-aguirre, 2016-04-01:
Not a bug, this is Windows intended design.

closed 2016-04-01 by rene-aguirre