rene-aguirre/pywinusb #35: `hid.core.show_hids()` raises UnicodeEncodeError with multibyte device name

menew, 2016-05-01:
because `hid.core.HidDevice.product_name` can be `unicode` object (by `c_wchar`) but `hid.core.HidDevice.__repr__` doesn't care about that.

I think `__repr__` should check object type, or simply ignore non-ascii characters by `encode('ascii', 'replace')` explicitly.

Could you consider fixing it?

pywinusb: 0.4.1
CPython: 2.7.8 (x64)
locale: ja

rene-aguirre, 2016-05-02:
@menew, have you seen the show_hids.py example? (https://github.com/rene-aguirre/pywinusb/blob/master/examples/show_hids.py), is this what you need.

Is the issue that you are reporting the fact that __repr__ does not return an unicode object?

Otherwise I don't think managing encodings in __repr__ (as os forcing ascii) is a good idea.

menew, 2016-05-02:
@rene-aguirre, thank you for reply!

> have you seen the show_hids.py example? (https://github.com/rene-aguirre/pywinusb/blob/master/examples/show_hids.py), is this what you need.

Yes, but passing `show_hids()` any encoders as output does not affect the situation.

For example, when I execute the following code:

    # coding: utf-8

    import sys
    import codecs
    import pywinusb.hid as hid

    output = codecs.getwriter('mbcs')(sys.stdout)
    hid.core.show_hids(output = output)

I got same UnicodeEncodeError as below, regardless of writer encoding:

    Found HID class devices!, writting details...

    Traceback (most recent call last):
      File "C:\Users\menew\Desktop\Documents\show_hids.py", line 8, in <module>
        hid.core.show_hids(output = output)
      File "build\bdist.win-amd64\egg\pywinusb\hid\core.py", line 1596, in show_hids
        device_name = str(dev)
    UnicodeEncodeError: 'ascii' codec can't encode characters in position 115-129: ordinal not in range(128)

And, if I edited `hid.core.HidDevice.__repr__` as this, then `show_hids()` shows report successfully without specifying output writer.

    def __repr__(self):
        # patched for python 2.x
        # (`"%s" % self.product_name` raises UnicodeEncodeError if `product_name` has non-ascii characters)
        return (u"HID device (vID=0x%04x, pID=0x%04x, v=0x%04x); %s; %s, " \
            u"Path: %s" % (self.vendor_id, self.product_id, self.version_number,\
            self.vendor_name, self.product_name, self.device_path)).encode('ascii', 'replace')

I think underlying problem is incompatibility that Python 2 demands `__repr__` to return non-unicode object whereas Python 3 demands unicode object. Because of the incompatibility, some version detection code or forcing ascii charset seem necessary to make `__repr__()` support both Python versions...

rene-aguirre, 2016-05-17:
"...without specifying output writer"

I think the library has to be encoding transparent. The decoding has to be managed by the application, not the library.

My question, about the "show_hids.py" example is that it actually fails with your device, "as is".

Just run it and let me know if this raises the Unicode error.

Also, when you refer to the japanese locale, is this the system locale (how did you set it?) or the USB device string encoding?

I'd like to reproduce your issue to look for alternatives, would be possible to get a dump of the device that includes string descriptors?

rene-aguirre, 2016-07-29:
@menew do you have a branch with your changes so issue #36 can be tested?

closed 2019-07-24 by rene-aguirre