rene-aguirre/pywinusb #36: one of my keyboards makes unicode error

K4zuki, 2016-07-28:
_may related with #35:_

Hello,
I would like to study the library, but facing problem at very first step.
Could you give any help please.
I can provide `lsusb -vv` result later (I am writing this from windows machine)

### condition
* Windows7 64bit
* Python 2.7.11 32bit
* 'FILCO' US ASCII USB keyboard

### what i did
* cloned the repo by usual `git clone`
* patched `examples/show_hids.py` to show python2 env and detail of error
```
diff --git a/examples/show_hids.py b/examples/show_hids.py
index 18c932f..ff92d9c 100644
--- a/examples/show_hids.py
+++ b/examples/show_hids.py
@@ -10,6 +10,7 @@ import pywinusb.hid as hid

 if __name__ == '__main__':
     if sys.version_info < (3,):
+        print "py2"
         import codecs
         output = codecs.getwriter('mbcs')(sys.stdout)
     else:
@@ -18,5 +19,5 @@ if __name__ == '__main__':
     try:
         hid.core.show_hids(output = output)
     except UnicodeEncodeError:
+        print sys.exc_info()
         print("\nError: Can't manage encodings on terminal, try to run the script on PyScripter or IDLE")
```
* run `python show_hids.py`

### result
```
$ python show_hids.py
py2
Found HID class devices!, writting details...
(<type 'exceptions.UnicodeEncodeError'>, UnicodeEncodeError('ascii', u'HID device (vID=0x04d9, pID=0x2011, v=0x0390); \u040
9; USB Keyboard, Path: \\\\?\\hid#vid_04d9&pid_2011&mi_01&col01#9&3766f47c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}',
47, 48, 'ordinal not in range(128)'), <traceback object at 0x025FA418>)
```

rene-aguirre, 2016-07-29:
Are you running your scripts on PyScripter, IDLE or any other environment that supports proper character encodings?

The error message there is set to reflect the limitations of the Windows command line, there are some encodings supported by your localized setting. But that's effectively limited by the OS (e.g. if you machine has Japanese locales, does not necessarily means you can see arabic characters on your terminal).

The issue here is not on PyWinUSB but how you show the information.

Usability question, if you are instructed to run the script (due not supported encoding of you device) on IDLE or PyScripter, why do you still expect to see not supported characters?

rene-aguirre, 2016-07-29:
@K4zuki , did you tested issue #35 solution on your machine? It looks it is exactly your problem. Please confirm so I can work on a python3 safe patch.

rene-aguirre, 2016-12-12:
Duplicate (lack of feedback to confirm otherwise)

closed 2016-12-12 by rene-aguirre