rene-aguirre/pywinusb #37: Py2Exe do not include pywinusb

timypik, 2016-07-29:
I've create *.exe binary by py2exe. 
But at the end of build i've got notes:
> The following modules appear to be missing
['pywinusb']

And when i've run *.exe i've got error, that 'pywinusb' not find.
But in another cases, like pyserial it works great.

rene-aguirre, 2016-07-29:
Can you provide some instructions and sample script to facilitate reviewing the issue?

linmonsv, 2016-09-14:
When I exec "python makefile.py py2exe"

That happend:

Error: Namespace packages not yet supported: Skipping package 'pywinusb'

  2 missing Modules
  ------------------
? pywinusb                            imported from __SCRIPT__
? readline                            imported from cmd, code, pdb
Building 'dist\show_hids.exe'.

pywinusb-0.4.1\examples\show_hids.py

-------------------------------------------------
from distutils.core import setup
import py2exe
import sys

py2exe_options = {
    "includes": ["sip"],
    "compressed": 1,
    "optimize": 2,
    "ascii": 0,
    "bundle_files": 1,
    }

setup(
  name = 'show_hids',
  version = '1.0',
  windows = [{"script":"show_hids.py", "icon_resources": [(1, "show_hids.ico")]}],
  zipfile = None,
  options = {'py2exe': py2exe_options}
  )

rene-aguirre, 2016-09-16:
This seems to be a py2exe limitation, can you try using pyinstaller as a workaround?

When initially taking the "pywinusb" namespace I started with the pywinusb.hid package, so anyone or myself could extend it or split it on new packages (e.g. pywinusb.pnp).

This never happened, so going to a monolithic package is feasible.

Anyway, using package namespaces was considered a "good practice" when I started this package, and important dependencies you might end up requiring would still be designed to be deployed as a namespace.

Hence, using a binary installer that supports package namespaces is the best way to go (or someone has to provide a package to py2exe to support them).

linmonsv, 2016-09-18:
Thank you for your help , the problem had been solved when I try to using pyinstaller !

closed 2016-12-12 by rene-aguirre