rene-aguirre/pywinusb #44: Can't find any devices

idanstark42, 2017-04-13:
I'm having the same problem.
I try to run the show_hids.py example, and I get "There's not any non system HID class device available"
When I looked in USBDeview.exe, I saw that the service my device uses is not hidUsb, but rather WinUsb, also it doesn't show in the device manager under Human Interface Device, but rather under Universal Serial Bus devices, but the device class is HID, so I don't know how should I procees from here.
 Is there any way to move it so that windows will refer to it as a HID device?

Thanks alot!

rene-aguirre, 2017-04-14:
@idanstark42 HID class devices are "hard" configured by the USB descriptors.

This is a hardware configuration and capabilities restriction.

One way to work around this, would be to have an adaptor intermediate device, which still sounds to much and specific for a given device. If this is your device (or you have the choice to modify the firmware), I'd rather suggest to change its USB descriptors and operations. Still not straightforward for most people though.

closed 2017-04-14 by rene-aguirre

idanstark42, 2017-04-14:
Hi! Thank you for replying. I found the device, thank you! I changed the
used driver through *device manager *from winusb to usbhid, and now the
system finds it. Thank you for your help!

rene-aguirre, 2017-04-15:
@idanstark42 interesting... Why do you think this happened? Did you manually install your device initially or had any "filter driver" dealing with it?

idanstark42, 2017-04-15:
Maybe. I did play a bit too much probably with the drivers, trying to use
libusb to make it work.

idanstark42, 2017-04-16:
Hi, I was wandering if you could help me out with another problem I have.
My device is a PS/2 mouse but it seems to only return to butttonUsage items
in the first report, and then not giving any other reports.
Also I looked at the report in debugger and it shows
that _HidReport__raw_report_size=2 but _HidReport__raw_data=None.
Do you know why this could happen?
thank you a lot

Idan

rene-aguirre, 2017-04-16:
That looks like a valid descriptor.

In this case you only have a single input report, so your device is instructing the HID parser to skip any report Id.

This is kind of a common descriptor structure for very simple devices, likely these are implemented in very resource constrained small SoC silicon devices.

idanstark42, 2017-04-16:
I see. Do you have any idea about the reports not comming in? Here's is my
code:

from pywinusb import hid
from time import sleep

device = hid.HidDeviceFilter(vendor_id=0x13ba,
product_id=0x0018).get_devices()[0]

def sample_handler(raw):
    print("data: {0}".format(raw))

try:
    device.open()

    device.set_raw_data_handler(sample_handler)

    while True:
        sleep(0.1)
finally:
    print("closing")
    device.close()

It's pretty straight forward, and yet the sample_handler is never called.

Thanks again for your help!

Idan

rene-aguirre, 2017-04-16:
There is a big restriction for trying to use HID input devices with PyWinUSB: Windows would "own" the device, so it's just not possible to share access.

If you can actually find your device is because it might expose non input devices usage pages (e.g. LED usage items) and only output or feature reports are available.

Makes sense?

Anyway, this is not a dead end if you still want to capture your device input. Here is where a USB comes handy, but honestly for this you'd might be better off with libusb et. al. kind of libraries.

PyWinUSB "just works", no need for special drivers or config, for regular non input devices, but for keyboards, mice and some game input devices you might actually do it better using other Windows API access calls or events, which actually some are kind of standard events.

idanstark42, 2017-04-16:
Ok, thank you.

I think I got it.

I did use libusb before, and got an empty report there, but I will try
again.

Thank you so much for helping, this is really not taken for granted.

Idan