rene-aguirre/pywinusb #45: pywinusb with pyinstaller

tanyanghan, 2017-11-03:
Hi there,

I'm trying to package up my python app that uses pywinusb with pyinstaller. 

I wrote a little Test.py:

```
import pywinusb.hid as hid

all_devices = hid.find_all_hid_devices()

if not all_devices:
    print("No devices found")

for device in all_devices:
    print(device)
```
This works fine and I get a print out of a list of devices like:
```
HID device (vID=0x04f3, pID=0x20d0, v=0x1111); ELAN; Touchscreen, Path: \\?\hid#vid_04f3&pid_20d0&col04#6&11e83ff1&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
```
 
When I then try to package it up using pyinstaller, it prints out `No devices found`.

If I then add a debug print out line to `find_all_hid_devices()` in  `core.py`

```
            # add device to results, if not protected
            if hid_device.vendor_id:
                results.append(hid_device)
            else:
                print("Not appended: " + str(hid_device))
```
I get this from the pyinstaller version:
```
Not appended: HID device (vID=0x0000, pID=0x0000, v=0x0000); ; , Path: \\?\hid#vid_04f3&pid_20d0&col04#6&11e83ff1&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
```

So it looks like it's finding the device paths, but somehow the vID and pID are zero. Any ideas on what could be missing from the pyinstaller package?

rene-aguirre, 2017-11-03:
Hello @tanyanghan , could you run your binary in an Administrator shell?

I think the OS might be protecting the USB access to an un-trusted executable.

If this works on a admin console, either you'd need to change the binary properties manually, or you might need to add some sort of trusted certificate (check pyinstaller docs).

tanyanghan, 2017-11-05:
Hi @rene-aguirre, thanks for your response. I have tried running it as under an admin console, but it didn't make a difference. I also tried right clicking on the exe in Explorer and running the exe from Explorer as admin but it still didn't work.

tanyanghan, 2017-11-08:
I've signed the exe with a self-signed certificate which I added to the Trusted Root Certification Authority on my PC, but the result is still the same with the vID and pID all returning as zeroes. The signing appears to have worked because the properties of the exe in Explorer now shows `This digital signature is OK.`

tanyanghan, 2017-11-08:
Further digging, `winapi.CreateFile()` appears to be returning a valid `h_hid` handle and calling `ctypes.GetLastError()` returns `0 - ERROR_SUCCESS`.

BUT when it tries to read the attributes `hid_dll.HidD_GetAttributes(h_hid, byref(hidd_attributes))`, it fails and `ctypes.GetLastError()` returns `6 - ERROR_INVALID_HANDLE`.

rene-aguirre, 2017-11-08:
@tanyanghan Could be an issue with your registry? Example: http://digital.ni.com/public.nsf/allkb/1D52B53149DD637686256CAA00565E47

tanyanghan, 2017-11-09:
I tried it, and it didn't seem to have made a difference.

tanyanghan, 2017-11-10:
Here's the problem. I was using Python x86-64 and it didn't work after pyinstaller packaged it up.
`Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:25:58) [MSC v.1500 64 bit (AMD64)] on win32`

It works with Python x86 (32-bit) after pyinstaller packages it up!
`Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:19:30) [MSC v.1500 32 bit (Intel)] on win32`

rene-aguirre, 2017-11-18:
Thanks for let us know @tanyanghan !

closed 2017-11-18 by rene-aguirre