rene-aguirre/pywinusb #47: sending output report

mfriedmann2, 2017-12-01:
Hello I am new to HID Communication Protocols and I only had previous I2C protocol experience and I am having some difficulties with just sending simple hexadecimals patches and receiving them with fixed length of bytes (raw_data).

I want to send the fixed length of 7 bytes to the device and I am receiving an error since the parameter in the send_output_report is not correct. Ithink it may be due to the data type. It must be sent in hexadecimals values.. what should I do to solve this issue?

This is the code I've got:

    filter = hid.HidDeviceFilter(vendor_id = 0x1234, product_id = 0x4321)
    dev = filter.get_devices()
    aqm=dev[0]
    aqm.open()
    buffer = [(c_ubyte(), 0xD3), (c_ubyte(),0x3D), (c_ubyte(), 0x07), (c_ubyte(),0x03), (c_ubyte(),0x00), (c_ubyte(),0xE2), (c_ubyte(),0x2E)]
    print(buffer)
    print(int(aqm.hid_handle))
    #aqm.send_output_report(buffer)
    aqm.close()

mcuee, 2018-04-28:
Ref: this may help you.
https://sourceforge.net/p/pyusb/mailman/message/31188626/

rene-aguirre, 2018-04-29:
This is no a library bug. :-)

As the page you pointed out suggest, try to be more familiar with USB HID.

> It must be sent in hexadecimals values..
That's how reports work.

Other possibility is that the data might be required to be sent in a feature report.

If you have a vendor app working, try to set a USB filter to know the reports you need to send.

closed 2018-04-29 by rene-aguirre