rene-aguirre/pywinusb #49: Can the raw_data read a HID-compitable mouse deivce ?

glucee, 2018-04-05:
I run the raw_data code, and it can recognise several devices (they are keyboard or touchpad), when I select anyone, none of them returns raw data. Is it the limitation of the pywinusb ? Or how can I get the raw data from the touchpad or similar device ? Thanks.

mcuee, 2018-04-08:
I think not if they are using different driver/API. pywinusb is for generic HID device, not for keyboard/mouse.

rene-aguirre, 2018-04-08:
Hello @glucee , that's actually a Windows OS restriction, HID input (keyboards, mice, some game devices) are restricted to direct application access as there are other system specific ways of dealing with those.

closed 2018-04-08 by rene-aguirre

mcuee, 2025-03-19:
One possibility: Raw Input
https://learn.microsoft.com/en-us/windows/win32/inputdev/raw-input