rene-aguirre/pywinusb #52: Report IN callback

NicoPy, 2018-08-22:
I have custom made HID devices with Vendor defined Features, IN and OUT reports.
Feature Get and Set are ok.
Report OUT is ok.
However, I have not been able to set a callback for Report IN transactions.
I use ```set_raw_data_handler()``` as a workaround but it is far from a good solution.

Did I missed something ?

rene-aguirre, 2018-08-23:
> Did I missed something?

:-) As everyone does.

If you are want a cleaner interface, decoupled from physical details use a proper HID descriptor that maps your report fields to usages.

For this check the telephone hook example: https://github.com/rene-aguirre/pywinusb/blob/master/examples/hook_button.py

The bare input report raw data handler was just a provision to deal with the fact that most HID devices don't use proper HID descriptors to describe the schema used, most of them don't even have reports on it.

What is your expectation for the input report? why not using the callback?

NicoPy, 2018-08-24:
> If you are want a cleaner interface, decoupled from physical details use a proper HID descriptor that maps your report fields to usages.

I don't map report fields to usages since my device is proprietary with lot of information transferred through end points. Most information fields are non standard.

> What is your expectation for the input report?

A way to register a callback for each Report IN. Each time a report IN message is received, the callback is called with message data.

> why not using the callback?
I do.