rene-aguirre/pywinusb #53: HidReport() class : bad self.__raw_report_size

NicoPy, 2018-08-22:
HidReport() class initialisation starts with the following :
```c
    def __init__(self, hid_object, report_type, report_id):
        hid_caps = hid_object.hid_caps
        if report_type == HidP_Input:
            self.__raw_report_size = hid_caps.input_report_byte_length
        elif report_type == HidP_Output:
            self.__raw_report_size = hid_caps.output_report_byte_length
        elif report_type == HidP_Feature:
            self.__raw_report_size = hid_caps.feature_report_byte_length
        else:
            raise HIDError("Unsupported report type")
```
```self.__raw_report_size``` is badly set to one of :
- hid_caps.input_report_byte_length 
- hid_caps.output_report_byte_length 
- hid_caps.feature_report_byte_length.

These values are the max of each report type.

For example, if there are 2 Report IN, one of size 4 and one of size 8, all HidReport() of type HidP_Input will have ```self.__raw_report_size``` set to 8. This is not correct.

rene-aguirre, 2018-08-23:
@NicoPy 
In your scenario if you have 2 input reports those should be declared in the HID descriptor.

The Windows HID API will pass your descriptor and provide individual details.

This is where there might be a some confusion: This library creates a helper instance of HidReport class for each exposed report in the HID descriptor.

If you support other transactions with an implicit descriptor interface you might need to manage your self the raw data.

NicoPy, 2018-08-24:
I modified ```HidReport()``` source code to get access to ```self.__raw_report_size``` through a ```report_size``` property.
I then display ```report_size``` for each report. I get the following :
FEATURE 1 5
FEATURE 2 5
FEATURE 3 5
FEATURE 4 5
FEATURE 5 5
FEATURE 6 5
IN 10 49
IN 11 49
IN 12 49
IN 13 49
IN 14 49
IN 15 49
IN 16 49
IN 17 49
OUT 8 2
OUT 9 2

This is not correct. Each report shall have a different size. 
Here, all report sizes are the max of all reports of the same type.

When receiving data in raw report callback, data passed to the callback should be the message content. This is not the case, data passed to the callback is the message content plus some data whose length is variable and make the whole data length equal to what is displayed above for a report type. 

Hmmm, this might not be very clear.
For example, report ID 10 is declared to be 10 bytes in length. When a report ID 10 message is received, the callback is called with 49 bytes (10 bytes of the message plus 39 bytes).

This makes the user code harder to debug : message length can't be checked, accessing data outside message content without error...

rene-aguirre, 2018-08-24:
Check this info: https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/hidpi/ns-hidpi-_hidp_caps

It seems that the only way to report different report sizes is to separate each report in a different collection, otherwise Windows might be just giving the maximum size value.

NicoPy, 2018-08-27:
I thought there is a way to know each report size as described in configuration descriptor.
Thanks for your support.

closed 2018-08-27 by NicoPy