rene-aguirre/pywinusb #56: Freeze when closing handle from __del__ during python exit

Ariakenom, 2019-06-27:
I tried to add a `__del__ `method for cleanup but it freezes if python exits while the object is alive.

`Python 3.6.3 (v3.6.3:2c5fed8, Oct  3 2017, 17:26:49) [MSC v.1900 32 bit (Intel)] on win32`
`pywinusb 0.4.2`
```
import pywinusb.hid

class C:
  def __init__(self,h):
    self.h = h
  def __del__(self):
    print("closing")
    self.h.close()
    print("closed")

h = pywinusb.hid.HidDeviceFilter().get_devices()[0]
h.open()
c = C(h)
# del c  # works if do this before python exits
```

rene-aguirre, 2019-07-24:
This is a known Python behavior. Example alternative: https://www.quora.com/Python-My-class-defines-del__-but-it-is-not-called-when-I-delete-the-object

Ariakenom, 2019-07-24:
The problem wasn't that the `del `method doesn't run, that would be ok behavior. It was that the python process doesn't exit, it freezes.

rene-aguirre, 2019-07-26:
Makes sense.

mcuee, 2021-08-12:
No issue here under Windows 10 with latest git.

```
(py39venv) C:\work\hid\pywinusb [master ≡ +1 ~0 -0 !]> python
Python 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:26:21) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pywinusb.hid
>>> class C:
...   def __init__(self,h):
...     self.h = h
...   def __del__(self):
...     print("closing")
...     self.h.close()
...     print("closed")
...
>>> h = pywinusb.hid.HidDeviceFilter().get_devices()[0]
>>> h.open()
>>> c = C(h)
>>> del c
closing
closed
```

Ariakenom, 2021-08-12:
`del c` is commented in the example

mcuee, 2021-08-13:
> `del c` is commented in the example

I know. But please look at my run log, I have the 'del c' in the run log.

Ariakenom, 2021-08-13:
It's not supposed to be in the run log.

That's why the statement is commented out. With a comment that says that it works as expected if we run the statement. But without it it does not work as expected and we get the freeze.

mcuee, 2021-08-14:
Thanks for the clarification. But it does not freeze for me.
```

(py39venv) C:\work\python> python
Python 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:26:21) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pywinusb.hid
>>> class C:
...   def __init__(self,h):
...     self.h = h
...   def __del__(self):
...     print("closing")
...     self.h.close()
...     print("closed")
...
>>> h = pywinusb.hid.HidDeviceFilter().get_devices()[0]
>>> h
HID device (vID=0x046d, pID=0xc534, v=0x2901); Logitech; USB Receiver, Path: \\?\hid#vid_046d&pid_c534&mi_01&col02#7&383a3a17&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
>>> h.open()
>>> c = C(h)
>>> c
<__main__.C object at 0x0000016ECD850F10>
>>> exit()
closing
closed
```

mcuee, 2021-08-14:
Without using the interactive session:

```
(py39venv) C:\work\python> cat .\test_pywinusb.py
import pywinusb.hid

class C:
  def __init__(self,h):
    self.h = h
  def __del__(self):
    print("closing")
    self.h.close()
    print("closed")

h = pywinusb.hid.HidDeviceFilter().get_devices()[0]
h.open()
c = C(h)

(py39venv) C:\work\python> python .\test_pywinusb.py
closing
closed
```

mcuee, 2021-08-23:
@Ariakenom Maybe you want to try again with more recent Python version to see if there is a real issue or not. Thanks.

mcuee, 2021-08-23:
I can reproduce the issue with Python 3.6.8 64bit.

```

(py36venv) C:\work\python> python
Python 3.6.8 (tags/v3.6.8:3c6b436a57, Dec 24 2018, 00:16:47) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.

(py36venv) C:\work\python> python .\test_pywinusb.py
closing
<hang>
```