rene-aguirre/pywinusb #61: cx_Freeze can't find module pywinusb

JulienCochuyt, 2020-04-07:
When trying to cx_Freeze an app using pywinusb 0.4.2 installed with pip, the build fails unable for find module pywinusb.
As described [here](https://sourceforge.net/p/cx-freeze/mailman/message/30998577/), it appears that __init__.py is missing from the installed root package.

Manually- adding an empty __init__.py to Lib/site-packages/pywinusb/ solves the issue and cx_Freeze builds just fine.

rene-aguirre, 2020-04-08:
Try with `pywinusb.hid` not `pywinusb`. That's the only package namespace implemented.

JulienCochuyt, 2020-04-08:
@rene-aguirre, thanks for the quick reply.
> Try with `pywinusb.hid` not `pywinusb`. That's the only package namespace implemented.

My bad.
Adding `pywinusb.hid` to the package list does not help as cx_Freeze tries (and fails) to walk up the package tree.
However cx_Freeze 4.3.2 added a `namespace_packages` option list which accepts `pywinusb` just fine.

closed 2020-04-08 by JulienCochuyt