rene-aguirre/pywinusb #65: Running show_hids.py - A USB HID not showing in the list

ADAMTER, 2020-07-14:
I have a USB barcode scanner connected and it can scan, it shows in the device manager, but does not show in the list of HIDs when running show_hids.py or raw_data.py.  Do you know why this might be? I can see the vendor ID and Product ID in the Event logs -- showing it was started and configured. Any help in understanding would be appreciated. thanks

mcuee, 2021-08-12:
It might be recognized by the system as a mouse or keyboard and then the HID API may not be able to find it.

mcuee, 2021-08-12:
You may try HIDAPI and cython-hidapi to see if that helps.

mcuee, 2021-08-12:
For example, here is the output under my system from pywinusb. You can see that mouse/keyboard are not listed.

```
Found HID class devices!, writting details...
HID device (vID=0x046d, pID=0xc534, v=0x2901); Logitech; USB Receiver, Path: \\?\hid#vid_046d&pid_c534&mi_01&col02#7&383a3a17&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
...
HID device (vID=0x046d, pID=0xc534, v=0x2901); Logitech; USB Receiver, Path: \\?\hid#vid_046d&pid_c534&mi_01&col03#7&383a3a17&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
...
HID device (vID=0x046d, pID=0xc534, v=0x2901); Logitech; USB Receiver, Path: \\?\hid#vid_046d&pid_c534&mi_01&col04#7&383a3a17&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
...
HID device (vID=0x8087, pID=0x0a1e, v=0x0200); Unknown manufacturer; @input.inf,%hid_wireless_radio_controls%;HID-compliant wireless radio controls, Path: \\?\hid#intc816&col01#3&36a7043c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
...
HID device (vID=0x046d, pID=0xc534, v=0x2901); Logitech; USB Receiver, Path: \\?\hid#vid_046d&pid_c534&mi_01&col05#7&383a3a17&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}
...
HID device (vID=0x8087, pID=0x0a1e, v=0x0200); Unknown manufacturer; @buttonconverter.inf,%btnconv.device%;Portable Device Control device, Path: \\?\hid#intc816&col02#3&36a7043c&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
...
HID device (vID=0x413c, pID=0xb06e, v=0x0101); Unknown manufacturer; @input.inf,%hid_device_vendor_defined_range%;HID-compliant vendor-defined device, Path: \\?\hid#vid_413c&pid_b06e#c&37ff1248&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
...
HID device (vID=0x047f, pID=0xc056, v=0x0210); Plantronics; Plantronics Blackwire 3220 Series, Path: \\?\hid#vid_047f&pid_c056&mi_03&col01#f&39e6f119&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
...
```

mcuee, 2021-08-12:
Output from cython-hidapi.

```
(py39venv) C:\work\hid\cython-hidapi [master ≡]> python
Python 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:26:21) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import hid
>>> import hid
>>>
>>> for device_dict in hid.enumerate():
...     keys = list(device_dict.keys())
...     keys.sort()
...     for key in keys:
...         print("%s : %s" % (key, device_dict[key]))
...     print()
...
interface_number : 1
manufacturer_string : Logitech
path : b'\\\\?\\hid#vid_046d&pid_c534&mi_01&col01#7&383a3a17&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}'
product_id : 50484
product_string : USB Receiver
release_number : 10497
serial_number :
usage : 2
usage_page : 1
vendor_id : 1133

interface_number : 1
manufacturer_string : Logitech
path : b'\\\\?\\hid#vid_046d&pid_c534&mi_01&col02#7&383a3a17&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}'
product_id : 50484
product_string : USB Receiver
release_number : 10497
serial_number :
usage : 1
usage_page : 12
vendor_id : 1133

interface_number : 1
manufacturer_string : Logitech
path : b'\\\\?\\hid#vid_046d&pid_c534&mi_01&col03#7&383a3a17&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}'
product_id : 50484
product_string : USB Receiver
release_number : 10497
serial_number :
usage : 128
usage_page : 1
vendor_id : 1133

interface_number : 1
manufacturer_string : Logitech
path : b'\\\\?\\hid#vid_046d&pid_c534&mi_01&col04#7&383a3a17&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}'
product_id : 50484
product_string : USB Receiver
release_number : 10497
serial_number :
usage : 1
usage_page : 65280
vendor_id : 1133

interface_number : -1
manufacturer_string : Microsoft
path : b'\\\\?\\hid#dell091a&col01#5&99b72d3&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}'
product_id : 4639
product_string : HIDI2C Device
release_number : 1037
serial_number : 9999
usage : 2
usage_page : 1
vendor_id : 1160
[... 196 lines cut ...]
usage_page : 13
vendor_id : 1160

interface_number : -1
manufacturer_string :
path : b'\\\\?\\hid#vid_413c&pid_b06f#d&3624b04c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}'
product_id : 45167
product_string :
release_number : 257
serial_number :
usage : 218
usage_page : 65498
vendor_id : 16700

interface_number : -1
manufacturer_string :
path : b'\\\\?\\hid#converteddevice&col01#5&379854aa&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd'
product_id : 0
product_string :
release_number : 0
serial_number :
usage : 6
usage_page : 1
vendor_id : 1118
```

mcuee, 2021-08-12:
Ref: hidapi change which adds support for keyboard and mouse
https://github.com/libusb/hidapi/commit/76d4ffd8538e40e95f11c139a146afa29be093c7#diff-bdce61d76faf6ea305f64cd6c5dc7d26887efc0ef63a73c2b77f1da09d095aa1

mcuee, 2021-08-12:
@rene-aguirre Do you like this idea? I understand that there is a limitation for keyboard/mouse under Windows, and that is why libusb HID backend does not support keyboard/mouse. But after testing out HIDAPI, I think it is actually useful.

giohappy, 2022-03-11:
Unofrtunately the limitation reported by @mcuee also affects smple HIDs built with Arduino Micro (Atmel AVR Atmega32u4).
It would be great to be able to parse these devices too.

I suspect that the first problem is that pywinusb implements only `HidD_GetPreparsedData`, which is only available for HIDClass devices. Myabe [GetRawInputDeviceInfo](url) should be used for simpler dvices

giohappy, 2022-03-11:
Well, apparently the issue is that for security reasons Windows doesn't allow to access mouse and keybord HIDs.
Might this be the reason pywinusb cannot access mouse and keyboard devices?

https://github.com/signal11/hidapi/issues/247#issuecomment-161152387

rene-aguirre, 2022-03-16:
by design, `pywinusb` doesn't use a helper kernel driver only the user space HID API

As you pointed out, HID API excludes r/w access to input and gaming devices, we can only peek to the descriptors using the HW enumeration APIs.

closed 2022-03-16 by rene-aguirre