rene-aguirre/pywinusb #75: PNP for tkinter

IFXGAWU, 2022-05-03:
Hello, 
I tried the pnp example for wx and it works fine for me. 
Is there maybe an example available for tkinter available?
tkinter does not provide a handle, do you have some hints for me how to implement pnp for tkinter?
Thanks in advance,
GW

rene-aguirre, 2022-05-03:
It seems there is a way to get the HWND, this looks like a good starting point: https://web.archive.org/web/20200731093951id_/http://effbot.org/tkinterbook/wm.htm#Tkinter.Wm.frame-method

IFXGAWU, 2022-05-05:
Thanks for your very fast response!
Could you please give me an additional helping hand?

 class MyFrame(ttk.Frame,hid.HidPnPWindowMixin):
    # a device so we could easily discriminate wich devices to look at
    my_hid_target = hid.HidDeviceFilter(vendor_id = target_vendor_id, product_id = target_product_id)

    def __init__(self, parent):
        ttk.WmFrame.__init__(self,parent) #tkinter Frame has only 2 parameters
        # wx.Frame.__init__(self,parent,-1,"Re-plug your USB HID device, watch the command window!...")
        #             Frame(parent, id=ID_ANY, title="", pos=DefaultPosition, size=DefaultSize, style=DEFAULT_FRAME_STYLE, name=FrameNameStr)
        hid.HidPnPWindowMixin.__init__(self, self.GetHandle())
        ttk.EVT_CLOSE(self, self.on_close)

Result:
AttributeError: module 'tkinter.ttk' has no attribute 'WmFrame'

Do you have the time to check what's going wrong?
Or do I have to adapt the "HidPnPWindowMixin" for tkinter?
Thanks & BR

rene-aguirre, 2022-07-12:
WMFrame seems to be legacy.

I check in a REPL session that a tkinter.Frame exposes a `winfo_id()` method, google shows this will be the frame HWND in Windows

IFXGAWU, 2022-07-13:
Hello Rene,
It looks like that's how it works.

`#
"""
Plug and Play example

This script requires wxPython, but it could be easily (really!) changed
to work with any GUI working on windows, just make you you pass your frame window
handler to the HidPnPWindowMixin.__init__ initialization function.

A hook will be inserted on the message handler so the window could be used as
a target for PnP events, for now this are HID class wise, this means you'll have
to test if your device 'plug' status has changed.
"""
from tkinter import *
from tkinter import ttk
import tkinter as tk
from tkinter import messagebox
from tkinter import filedialog
import tkinter as tk

import pywinusb.hid as hid
import sys
from time import sleep

# feel free to test
target_vendor_id = 0x640
target_product_id = 0xA10

class MyFrame(ttk.Frame,hid.HidPnPWindowMixin):
    # a device so we could easily discriminate wich devices to look at
    my_hid_target = hid.HidDeviceFilter(vendor_id = target_vendor_id, product_id = target_product_id)

    def __init__(self, parent):
        ttk.Frame.__init__(self,parent)
        # wx.Frame.__init__(self,parent,-1,"Re-plug your USB HID device, watch the command window!...")
        #             Frame(parent, id=ID_ANY, title="", pos=DefaultPosition, size=DefaultSize, style=DEFAULT_FRAME_STYLE, name=FrameNameStr)
        #hid.HidPnPWindowMixin.__init__(self, self.GetHandle())
        hid.HidPnPWindowMixin.__init__(self, self.winfo_id()) #or try winfo_id()
        #ttk.EVT_CLOSE(self, self.on_close)
  
        
        self.device = None #no hid device... yet

        # kick the pnp engine
        self.on_hid_pnp()

    def on_hid_pnp(self, hid_event = None):
        """This function will be called on per class event changes, so we need
        to test if our device has being connected or is just gone"""
        # keep old reference for UI updates
        old_device = self.device

        if hid_event:
            print("Hey, a hid device just %s!" % hid_event)

        if hid_event == "connected":
            # test if our device is available
            if self.device:
                # see, at this point we could detect multiple devices!
                # but... we only want just one

                pass
            else:
                self.test_for_connection()
        elif hid_event == "disconnected":
            # the hid object is automatically closed on disconnection we just
            # test if still is plugged (important as the object might be
            # closing)
[... 67 lines cut ...]
        event.Skip()
        if self.device:
            self.device.close()

if __name__ == "__main__":
    buffer = [0x00] * 65
    buffer[0] = 0x00
    
    
    root = Tk()

    #frame = Frame(root)
    #frame.pack()

    #frame = MyFrame(None)
    frame = MyFrame(root)
    frame.pack()
    
    root.title("HID test pywinusb with TKINTER 2022/05/08")
    root.geometry("1800x1000")
    root.geometry("1920x1000+-8+0")
    
    #frame.Show()   
    
    root.mainloop()`