rene-aguirre/pywinusb #78: HID RFID Tag Reading Issue

hobik, 2022-11-09:
I've Rfid reader, when I read the rfid card, tag is writes in txt file as a keyboard. I know vid and pid for this product. And I am using Windows10.

I'm trying to work with below codes, print("scan_hiddevice: found %s", device ) working correctly and gives us:

`scan_hiddevice: found %s HID device (vID=0xffff, pID=0x0035, v=0x0100); IC Reader; UserDefHid, Path: \\?\hid#vid_ffff&pid_0035&mi_01#7&110b018b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}`
I expect rx_handler give me tag but it couldnt give anything when I read my rifd card. Where is my fault ? How can I get rfid tag with python ?

```
import pywinusb.hid as hid

DEVICE_DEFAULT_VID = 0xffff
DEVICE_DEFAULT_PID = 0x0035
# handler called when a report is received
def rx_handler(data):
    print ('recv: ', data)

def scan_hiddevice():
    """ Scans for and returns the HID device. """
    devices = hid.HidDeviceFilter( vendor_id = DEVICE_DEFAULT_VID, product_id = DEVICE_DEFAULT_PID ).get_devices()
    if not devices:
        print ("scan_hiddevice: No device connected.")
        return None
    else:
        device = devices[0]
        print("scan_hiddevice: found %s", device )
        print(device)
        return device    
    return None

def setup_hiddevice():
    """Creates a new HID device, opens it and attaches a receive data handler"""
    hid_device = scan_hiddevice()
    print("here")
    hid_device.set_raw_data_handler(rx_handler)
    hid_device.open()

    print(hid_device)
    return hid_device

def main(verbose=True):
    hid_device = setup_hiddevice()

    while (True):
        #wait for data
        1 == 1

main()
```
I am using this device: [https://www.amazon.com.tr/125kHz-RFID-Okuyucu-Modül-EM4100/dp/B07HMR9ZVX/ref=asc_df_B07HMR9ZVX/?tag=trshpngglede-21&linkCode=df0&hvadid=510131658719&hvpos=&hvnetw=g&hvrand=2287751725591225714&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1012764&hvtargid=pla-988089364799&psc=1](https://www.amazon.com.tr/125kHz-RFID-Okuyucu-Mod%C3%BCl-EM4100/dp/B07HMR9ZVX/ref=asc_df_B07HMR9ZVX/?tag=trshpngglede-21&linkCode=df0&hvadid=510131658719&hvpos=&hvnetw=g&hvrand=2287751725591225714&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1012764&hvtargid=pla-988089364799&psc=1)

When this program running, when I read rfid card output writting cursor area but I couldnt get rx_handler

rene-aguirre, 2022-11-10:
check the usb descriptor, look online for a tool, if it reports as a system input device (keyboard, game controller, mouse), you need to use other libraries (or just standard python `input()` calls).

tissi-2, 2022-12-21:
```
import pywinusb.hid as hid
import getpass
import time

DEVICE_DEFAULT_VID = 0xffff
DEVICE_DEFAULT_PID = 0x0035

def scan_hiddevice():
    """ Scans for and returns the HID device. """
    devices = hid.HidDeviceFilter( vendor_id = DEVICE_DEFAULT_VID, product_id = DEVICE_DEFAULT_PID ).get_devices()
    if not devices:
        print ("scan_hiddevice: No device connected.")
        return None
    else:
        device = devices[0]
        print("scan_hiddevice: found %s", device )
        return device    
    return None

def setup_hiddevice():
    """Creates a new HID device, opens it and attaches a receive data handler"""
    hid_device = scan_hiddevice()
    print("here")
    hid_device.open()
    return hid_device

def main(verbose=True):
    hid_device = setup_hiddevice()

    while (True):
        keys = getpass.getpass("Swipe Card")
        print (keys)
        time.sleep(1)

main()
```
Maybe this helps...

arawn45, 2023-07-08:
thank you for your file, which helped me a lot too.

I would like the input to be done in the background to insert the rfid reading into a sql database

is there a way to make it work in the background?

Thanks for your help
Christopher

*******************
en francais :

merci pour votre fichier, qui m'a bien aidé aussi.

j'aimerai que la saisie se fasse en tache de fond pour inserer la lecture rfid dans une base sql

y a t il moyen de faire fonctionner en tache de fond?

merci pour votre aide
Christophe